mattytrentini/led_strip #5: led_strip_breadboard.fzz: Unable to find part

HowManyOliversAreThere, 2017-06-03:
Upon opening the led_strip_breadboard.fzz Fritzing file, an error pops up stating, "**Unable to find the following part:** 'Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals' **at** '/home/<USER>/Documents/fritzing-*/fritzing-parts/core/Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals.fzp' ". Error seen on Fritzing 0.9.3b.

mattytrentini, 2017-06-03:
Hmm, interesting - as far as I can tell that's a [standard part](https://github.com/fritzing/fritzing-parts/blob/master/core/Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals.fzp). I'm on 0.9.3b too! I'll investigate further tomorrow; if I can't work out what's going on I'll just replace that part with another...

mattytrentini, 2017-06-03:
By the way, here's an image of the breadboard:

[image]

HowManyOliversAreThere, 2017-06-05:
No terminal block for me - definitely seems to be referencing your local copy for some reason?

[image]

mattytrentini, 2017-06-06:
I just tried installing Fritzing on my work computer (Windows, whereas at home I'm on Ubuntu) and the board loaded fine, including the terminal block...hmm.

I'll try to find a moment to try to swap out to another component anyway.

HowManyOliversAreThere, 2017-06-06:
I'm on Windows 10, although it still seems strange that it's such a specific error - oh well, happy to write this off as something on my end!

closed 2018-03-10 by mattytrentini