mattytrentini/led_strip #7: main.py: LED strip driving functions should be brought out into own class

HowManyOliversAreThere, 2017-06-08:
Currently the LED strip driving calls are all nestled within the main.py file. It would be good to extract these calls / functions, neaten them up, and place them in a class so that multiple LED strips can be driven at once. Fading and debouncing logic will have to be re-designed to a non-blocking implementation.

HowManyOliversAreThere, 2017-06-08:
Possible example of class usage:

```python
import led_strip
my_led_strip = led_strip(7) # LED strip driver connected to pin 7
my_led_strip.on() # turns theLED strip to full brightness
my_led_strip.pwm(70) # pulses the LED strip at 70% of the maximum PWM
print('PWM value: %i' % my_led_strip.getPwm()) # prints the current PWM value (0 for off, 100 for fully on)
my_led_strip.off() # turns the LED strip off
```

Could possibly also have `increment` and `decrement` functions that increase of decrease the pwm by a fixed amount, amount settable by a `setIncrements(incrementSize)` function

mattytrentini, 2017-06-08:
Agree with pulling the need to encapsulate the led strip functionality. And yes, some thought will need to be given to how to fade and debounce nicely, I'm leaning toward a design based around [asyncio](https://docs.python.org/3/library/asyncio.html). [This example](https://github.com/micropython/micropython/blob/master/examples/hwapi/soft_pwm_uasyncio.py) is close to what I'm thinking (though we'd use PWM support rather than do it manually). 

As for the interface, you've made a good start though a couple of changes have popped out:
```python
from machine import PWM, Pin 
import led_strip
my_led_strip = led_strip(machine.PWM(machine.Pin(7), freq=1000), max_duty=1023) # [1]
my_led_strip.on() # turns the LED strip to full brightness [2]
my_led_strip.brightness(70) # Set the strip to a percentage of maximum brightness [3]
print('Brightness: %i' % my_led_strip.brightness) # prints the current brightness value
my_led_strip.off() # turns the LED strip off [4]
```

1. I think we should just expect an object to be passed in that provides a 'duty' method rather than doing all the GPIO management internally. This makes it more composable and portable.
2. Just calls .duty(max_duty). Or .brightness(100).
3. Shouldn't really have 'pwm' in the interface. Probably needs to take a float
4. .duty(0)

But this is all off the top of my head, really need to experiment a little more...

HowManyOliversAreThere, 2018-01-26:
Class implementation has been resolved as at ef2ca578456f067787f2076fbe4f8cbc3fbb26fd

closed 2018-03-10 by mattytrentini