mattytrentini/led_strip #16: Fade needs to be smoother

mattytrentini, 2018-03-11:
There are obvious steps in changing the brightness when fading, it should be smoother. If we put a simple fade routine in a (non-async) loop:

```py
from machine import Pin, PWM
from time import sleep_ms

led=PWM(Pin(22))
for i in range(1024):
	led.duty(i)
	sleep_ms(5)
```

It's _very_ smooth. 

A few issues to consider:

- Need to time how long the async sleep was _actually_ asleep for
  - Just because we request x ms doesn't mean we will _exactly_ get it. 
  - Initial measurements look like it's typically a few ms later (ie async.sleep_ms gives a _minimum_ sleep time)
  - Adjust the brightness step by the _actual time_ since the last step
- Use a _rate of change_ to adjust brightness
  - Something like 0-100% brightness in 1.5 sec
- May be best to change brightness to a percentage
- Brightness appear to be non-linear
  - Fade routine can work with a linear percentage but we can introduce a function to return a non-linear percentage
- The last step may need some care
  - For example if we get _very_ close to the goal brightness on the second-last step we may sleep for too long
- Need to time what the minimum sleep time of async.sleep_ms is
  - Initial measurements look like it can go down to 5ms though the documentation suggests 10's of ms is more realistic
  - Even a 20ms sleep should be able to look smooth? Need to test...
- Remember that print statements can affect the timing dramatically!
  - Maybe put them in another async loop that's serviced infrequently

mattytrentini, 2018-03-12:
Tried a non-async 20ms step (1.5s 0-100%) fade and it's _very_ smooth:

```py
from machine import Pin, PWM
from time import sleep_ms
def fadein():
    led=PWM(Pin(12))
    for i in range(0, 1024, 13): # 13 = 1024/(1500ms/20ms)
        led.duty(i)
        sleep_ms(20)
```

seanlano, 2018-07-29:
On the point:

> Brightness appear to be non-linear

I found that using an exponential function gave a better-looking response - in my case in the WiLED project I had a notion of "brightness steps", i.e. the dimmer would change between say five distinct levels only, rather than the full range PWM values. Then I have a mapping between those brightness levels and an actual PWM value. I suppose if you wanted to do 100 "brightness steps" then those would effectively be a percentage to PWM mapping. 

The C++ code I have for it is:
```
float max_log10 = log10(led->getMaxPWM());
float steps_log10 = log10(NUM_DIM_STEPS-1);
float exponent = max_log10/steps_log10;
float result;
pwm_dim_levels[0] = 0;
for(int idx=1; idx<NUM_DIM_STEPS; idx++){
	result = pow(idx,exponent);
	pwm_dim_levels[idx] = int(result);
}
pwm_dim_levels[NUM_DIM_STEPS-1] = led->getMaxPWM();
```

The gist of it is though, suppose you want 8-bit values and 6 steps, you find the exponent with `exponent = log10(255)/log10(6) = 3.092638069` and then the actual value with `pwm = step ^ exponent`.

Then the values for each step are:
```
pwm_dim_levels[0] = 0
pwm_dim_levels[1] = 1
pwm_dim_levels[2] = 9
pwm_dim_levels[3] = 30
pwm_dim_levels[4] = 72
pwm_dim_levels[5] = 145
pwm_dim_levels[6] = 255
```

mattytrentini, 2018-07-29:
Thanks @seanlano!

I was had intended to insert a mapping shim method - pass a function a value from 0-100 and then have it return a non linear value from 0-100.  Something like what you've proposed ought to work well!

Looking forward to seeing your LED controller in the flesh... :)