mattytrentini/led_strip #18: Need better control over LED strips on startup

mattytrentini, 2018-03-11:
Currently when the button is pressed the LED's come on temporarily at full brightness then back to completely off.

When the rotary encoder is pressed it _allows_ the LED's to be lit but starts them at 0 brightness. Once the encoder is turned it enforces a minimum brightness.

What should happen is that when the application starts the LED's should - perhaps - turn on and off quickly but with a very dim brightness. Then, if the rotary encoder is pressed, the LED's should be lit to the minimum brightness.

HowManyOliversAreThere, 2018-04-28:
Commit 98503565952ea704eea19dd15764a2eda591c734 sets initial LED brightness to 511 (out of 1023)- this is not the minimum brightness specified, but better than the previous behaviour (of being in an illegal state).

HowManyOliversAreThere, 2018-04-28:
Commit 031a866 strives to ensure that the LED drivers are disabled as soon as possible after startup, however there is still a period upon startup where the lights are on fairly strongly. I believe this can be attributed to the start-up time of Micropython combined with a lack of pull-down resistors on the LED_DRIVE_x nets. This should be resolved in the next revision of hardware - a new Issue (#20) has been created specifically for this implementation.